mantidproject/mantid #24636: Explore picking interaction points in simple shapes in MonteCarloAbsorption

soininen, 2019-01-29:
`MonteCarloAbsorption` picks the interaction points within the sample sample by choosing a random point inside the bounding box and then checking if the point is inside the actual shape. This may need a lot of random points and can fail completely if the actual shape does not fill the bounding box and the void volume is relatively large.

The geometry of simple shapes (cuboid, cylinder, ...) is well known in Mantid. It might be possible to pick the random points directly within the shapes without the detour of bounding box. This would give a usable interaction point without drawing a large number of random numbers speeding up the simulation and preventing failures.

SpencerHowells, 2019-01-29:
We adopted the current method for cylinders because we were informed that in the original version the distribution was not random.

martyngigg, 2019-01-30:
@SpencerHowells the distribution is random (I think we fixed a bug around that a while ago).

@peterfpeterson has been looking into these things and we agree we can do better for the simple shapes. It would be good to touch base with the work he has done so far.

martyngigg, 2019-01-30:
I agree. I think having a basic shape for the hollow cylinder would be the best way to go. I'll have to dig through the code to find out how we would go about that. I remember adding one at the level of the XML but I think it got constructed from other shapes in the end.

peterfpeterson, 2019-01-30:
It is the difference between [parseCylinder](https://github.com/mantidproject/mantid/blob/master/Framework/Geometry/src/Objects/ShapeFactory.cpp#L413) and [parseHollowCylinder ](https://github.com/mantidproject/mantid/blob/master/Framework/Geometry/src/Objects/ShapeFactory.cpp#L528). For the way the rest of `Rasterize` works, it would need to be added to the [`ShapeInfo` enum](https://github.com/mantidproject/mantid/blob/master/Framework/Geometry/inc/MantidGeometry/Rendering/ShapeInfo.h#L30). Those are the shapes that could go through `Rasterize` most easily.

closed 2019-02-19 by peterfpeterson