mantidproject/mantid #24662: Indirect Data Analysis - EndX and StartX go back to default when switching between workspaces

robertapplin, 2019-01-31:
### Expected behavior
When you choose an endX and startX for a spectra, this should remain the endX and startX for that spectra unless you try change it again.

### Actual behavior
The endX and startX for a spectra will go back to the default value when you switch between the workspace you have selected.

### Steps to reproduce the behavior
1.`Interfaces`->`Indirect`->`Data Analysis`->`ConvFit` tab
2. Go to Multiple Input and add the workspaces below
3. For the iris workspace, adjust the startx and endx using the sliders on the miniplot.
4. Switch to the osiris workspace using the combo box above the miniplots
5. Switch back to the iris workspace. The startx and endx have changed back to the default values!

**Data Files**
[Data Files.zip](https://github.com/mantidproject/mantid/files/2817266/Data.Files.zip)

### Platforms affected
All

Mantid-Matthew, 2020-09-01:
@Harrietbrown Another issue to-do with startX and EndX not working in the table.

robertapplin, 2020-09-08:
I noticed during unscripted testing that this also happens on Elwin.

robertapplin, 2020-10-15:
This can no longer be replicated on the fitting tabs by me or @Harrietbrown . However, a similar problem still exists on Elwin

PR #29762 (2020-10-21): Remove resetting of the fit range on changing workspace in elwin
This PR removes the resetting of the fit ranges when changing the plotted workspace in elwin tab on indirect data analysis.

**Report to:** Spencer 

**To test:**

1. Open indiect data analysis on the Elwin Tab
2. Load data from multiple runs.
3. Change between ploted workspaces from the dropdown menu. The fitting integretion range should stay the same
4. Change the range yourself.
5. Change workspace, the range should remain as what you set it as.

Fixes #24662 

---

#### Reviewer ####

Please comment on the following ([full de [... cut]
closed 2020-10-21 by martyngigg