mantidproject/mantid #24709: Export HelpWindow to the Workbench

martyngigg, 2019-02-06:
The `HelpWindow` class is used to provide the offline help for MantidPlot. The same class should be able to be used for the workbench. It needs to be accesssible from Python so will require exporting and having wrapper modules provided.

peterfpeterson, 2019-03-12:
Would rewriting this in python be better? https://doc.qt.io/qtforpython/overviews/qthelp-framework.html

martyngigg, 2019-03-12:
It's an idea. Just watch out with that link as it's about the Qt projects new Python bindings and not `PyQt`. `PyQt5` does have QtHelp bindings though. @hsautessella can probably comment on this further.

peterfpeterson, 2019-03-12:
My recommendation is to try bringing up the help system in python against our existing `qhc` file and see what happens.

hsautessella, 2019-03-27:
Sorry for the delayed response, I had to remind myself exactly where I was up to before the release period. 

I have got the help menu running locally in Workbench via the `InterfaceManager`. The main issue I was having was (on Windows) Qt was looking in the wrong place for the executable and resource files that QtWebEngine needs ([detailed here](https://doc.qt.io/qt-5/qtwebengine-deploying.html#deploying-qt-webengine-processes)).

This is solvable by creating a `qt.conf` file in `external\src\ThirdParty\lib\python2.7\` that points Qt to the correct place So the `qt.conf` file simply contains: 
```
[Paths]
Prefix = ../qt5
```
I'm not familiar with the packaging process so I don't know if this raises some packaging issues or not. Or whether or not rewriting in Python would solve this.

PR #25505 (2019-04-29): Export help window to workbench
**Description of work.**
Enable offline help in Workbench.

For Windows this involved having CMake move QtWebEngine resource files to the package build and embedding a `qt.conf` file in Workbench's `resources.py` to direct Qt to those files. Embedding was necessary to avoid pointing MantidPlot to the wrong places.

**To test:**
- Open Workbench and open the help window (either via the help menu or using the F1 shortcut).
- Make sure:
    - help pages are displayed correctly
    - http links open in your web browser
    - links direct you to the correct page
     [... cut]
closed 2019-04-29 by rosswhitfield