mantidproject/mantid #24717: IntegrateMDHistoWorkspace crash

mantid-roman, 2019-02-07:
### Expected behavior
Works

### Actual behavior
Crashes

### Steps to reproduce the behavior
Run this script:
```
nx = 512
ny = 192
nbins = 62
ndata = nx * ny * nbins
signal = range(0,ndata)
err = range(0,ndata)

mdws = CreateMDHistoWorkspace(SignalInput=signal,
                              ErrorInput=err,
                              Dimensionality=3,
                              Extents='0,{nbins},0,{ny},0,{nx}'.format(nx=nx, ny=ny, nbins=nbins*2),
                              NumberOfBins='{nbins},{ny},{nx}'.format(nx=nx, ny=ny, nbins=nbins),
                              Names='A,B,C',
                              Units='a,b,c')

cut = IntegrateMDHistoWorkspace(InputWorkspace=mdws, P1Bin=[], P2Bin=[85, 95], P3Bin=[])
```

### Platforms affected
Probably all. Seen on windows.

The cause of the crash is a component of a multi-dimensional index in `MDHistoWorkspaceIterator` going out of range. The multidimensional index is used to calculate the linear index. It the latter goes out of range mantid crashes.

AndreiSavici, 2019-02-07:
works on Ubuntu 18.04, at least in ipython on something close to nightly

mantid-roman, 2019-02-08:
@AndreiSavici Run it a few times. It should crash. It did for me (Ubuntu 18.04, today's master).
To see where the problem is replace this function `Mantid::coord_t
MDHistoWorkspaceIterator::jumpToNearest(const VMD &fromLocation)` with this code:

```
Mantid::coord_t
MDHistoWorkspaceIterator::jumpToNearest(const VMD &fromLocation) {
  std::vector<size_t> indexes(m_nd);
  coord_t sqDiff = 0;
  bool bad = false;
  for (size_t d = 0; d < m_nd; ++d) {
    coord_t dExact = getDExact(fromLocation[d], m_origin[d], m_binWidth[d]);
    size_t dRound = std::lround(dExact); // Round to nearest bin edge.
    if (dRound >= m_indexMax[d]) {
      bad = true;
      std::cerr << "Bad " << dRound << ' ' << m_indexMax[d] << std::endl;
    }
    sqDiff += (dExact - coord_t(dRound)) * (dExact - coord_t(dRound)) *
              m_binWidth[d] * m_binWidth[d];
    indexes[d] = dRound;
  }

  size_t linearIndex =
      Utils::NestedForLoop::GetLinearIndex(m_nd, &indexes[0], m_indexMaker);
  if (bad) {
    std::cerr << linearIndex << ' ' << m_max << std::endl;
    //linearIndex = m_max - 1;
  }
  this->jumpTo(linearIndex);
  return std::sqrt(sqDiff);
}
```

mantid-roman, 2019-02-08:
If you add this assertion `TS_ASSERT(itOut->isWithinBounds(itOut->getLinearIndex()));` at the end of test `MDHistoWorkspaceIteratorTest::test_jump_to_nearest_1d()` it will fail.
There is some confusion in the iterator's code: sometimes the index numbers the bins, sometimes the bin boundaries.

PR #24726 (2019-02-14): IntegrateMDHistoWorkspace crash
**Description of work.**
Added some bounds checking for an md iterator.

**To test:**
Run the script from the issue a few times.

Fixes #24717

*This does not require release notes* because nobody new there was a bug there.

---

#### Reviewer ####

Please comment on the following ([full description](http://developer.mantidproject.org/IndividualTicketTesting/)):

##### Code Review #####

- Is the code of an acceptable quality?
- Does the code conform to the [coding standards](http://developer.mantidproject.org/Standards/)?
- Are the unit tests small and test the [... cut]
closed 2019-02-14 by gvardany