mantidproject/mantid #24730: Manage Directories Opens Multiple Times

TTitcombe, 2019-02-08:
**Original reporter:** @Pasarus 

### Expected behavior
Only able to open Directory Manager Once

### Actual behavior
Can open as many tabs as you want

### Steps to reproduce the behavior
1. Interfaces -> SANS -> SANS ISIS v2 experimental
2. In the top right corner, click manage directories multiple times
3. Multiple GUIs will appear

### Platforms affected
All

martyngigg, 2019-03-19:
I can reproduce this in MantidPlot but not in Workbench. In Workbench I get the dialog as modal so I can't interact with the rest of the application. Can you still reproduce this on the workbench?

TTitcombe, 2019-03-19:
I still see this on workbench if you go to SANS -> ISIS SANS and click the manage directories button near the top right of the gui

martyngigg, 2019-03-19:
Ah I see. I forgot SANS has its own button. The general one from the file menu in workbench is okay so I guess the SANS ones just needs to be set modal?

PR #25303 (2019-03-22): Stop Multiple Manage Directories Opening in SANS
**Description of work.**
This PR makes the manage directories widget accessible from SANS modal, so only one can be opened at a time.

**To test:**
1. Interfaces -> SANS -> ISIS SANS
2. In the top right hand corner, click the manage directories button.
3. Check that, not matter how many times you click it, only one window will open
4. In the add runs tab (left hand side), check this manage directories button

Fixes #24730 

*This does not require release notes* because **Small change to way we handle widget. Not raised by users**

---

#### Revi [... cut]
closed 2019-03-22 by DanNixon