mantidproject/mantid #24766: BASIS: Transfer ModeratorTzeroLinear parameters to reflection-specific parameter files

jmborr, 2019-02-13:
Current parameters `Moderator.TimeZero.Gradient` and `Moderator.TimeZero.Intercept` are stored in file `BASIS_Parameters.xml`. They must be transferred to files `BASIS_silicon_111_Parameters.xml` and `BASIS_silicon_311_Parameters.xml` because they are to become reflection dependent.

To-Do:
- [ ] Transfer ModeratorTzeroLinear parameters to reflection-specific parameter files
- [ ] Modify BASIS reduction algorithm(s) to enforce this change

SpencerHowells, 2019-02-14:
@jmborr If the default facility is set to SNS then interface Indirect>Data Reduction will appear with BASIS parameter files. All ready for an interface for Reduction!

jmborr, 2019-02-14:
@SpencerHowells I wish it was that easy 😄 
The new moderator in SNS has different parameters for delayed emission times, so now I have a set of parameters for old runs and a set for new runs. Furthermore, the parameters are different for the different silicon reflections. BASIS is using reflections from silicon 111, 333, and 311 planes.
Thus I have to put two set of parameters in each of the BASIS_Parameters_silicion_XXX.xml files, where XXX is the reflection.
I have algorithm BASISReduction that will take care of resolving and fetching the appropriate set of parameters.

SpencerHowells, 2019-02-14:
@jmborr I was thinking more of launching BASIS reduction from the interface. Our IPFs, as well as having 'efixed', have 'resolution'. The latter is the FWHM resolution used for default values in various interfaces eg elastic window.

jmborr, 2019-02-14:
@SpencerHowells I sure put all the parameters (`resolution`, `Workflow.SE-log`, ...) needed by the indirect *analysis* interface in BASIS_Parameters.xml and BASIS_Parameters_silicon_XXX.xml files. However, I prefer to have a separated algorithm for reduction. Maybe in the future I could include it in the reduction interface, I'm interested in trying the available corrections. 
By the way, I can't find normalization by monitor and by vanadium calibration file in the reduction interface.

SpencerHowells, 2019-02-14:
@jmborr We use ISISIndirectEnergyTransfer which performs a wavelength-dependent monitor normalisation.  The Reduction interface has a Calibration tab and creates a WS *_calib by integrating over the vanadium elastic peak. The EnergyTransfer tab has an option to use this file.
The absorption corrections work in wavelength & needs the detector angle for each spectrum. Our *_red files are in angle but it is possible to use *_sqw. The routines convert the Q back to an angle, but this will only be approximate.

jmborr, 2019-02-14:
@SpencerHowells I see, the reason I could not find them is that those tabs are not generated when one selects BASIS (see screenshot below). Most likely some missing configuration in the parameter file(s)

[image]

SpencerHowells, 2019-02-14:
@jmborr We assumed that each facility would create their own EnergyTransfer tab. So when facility is ISIS the tab is called ISIS EnergyTransfer ; if ILL it is ILL EnergyTransfer. You would create SNS EnergyTransfer or more correctly SNS S(q,w) and not have the S(Q,w) tab which we use to convert *_red to *_sqw.

jmborr, 2019-02-15:
@SpencerHowells thanks for letting me know. It's something I will have to look onto since this interface offers more options than the reduction algorithm that I have. For the moment, thought, I have to make sure the algorithm works after the changes to the SNS moderator and the new Data Acquisition Software installed in BASIS.

closed 2019-02-18 by martyngigg