mantidproject/mantid #24801: Assertion failure cancelling live data operation using Kafka listener

martyngigg, 2019-02-18:
### Expected behavior

A live data operation should be able to be cancelled cleanly.

### Actual behavior

While testing #24799 it was observed that cancelling `MonitorLiveData` fires a debug assertion regarding an empty shared_ptr. Terminal log:

```
ConfigService-[Information] Unable to locate directory at: /etc/mantid/instrument
ConfigService-[Information] This is Mantid version 3.13.20190218.859 revision g1b57e42
ConfigService-[Information] running on ndw1213 starting 2019-02-18T09:13Z
ConfigService-[Information] Properties file(s) loaded: /media/data1/source/github/mantidproject/mantid-builds/nightly-py2/bin/Mantid.properties, /etc/mantid.local.properties, /home/dmn58364/.mantid/Mantid.user.properties
ConfigService-[Information] Unable to locate directory at: /etc/mantid/instrument
FrameworkManager-[Notice] Welcome to Mantid 3.13.20190218.859
FrameworkManager-[Notice] Please cite: http://dx.doi.org/10.1016/j.nima.2014.07.029 and this release: http://dx.doi.org/10.5286/Software/Mantid
StartLiveData-[Notice] StartLiveData started
LoadLiveData-[Notice] Performing the Replace operation.
MonitorLiveData-[Notice] MonitorLiveData started
StartLiveData-[Notice] StartLiveData successful, Duration 0.20 seconds
python: /usr/include/boost/smart_ptr/shared_ptr.hpp:648: typename boost::detail::sp_member_access<T>::type boost::shared_ptr<T>::operator->() const [with T = Mantid::DataObjects::EventWorkspace; typename boost::detail::sp_member_access<T>::type = Mantid::DataObjects::EventWorkspace*]: Assertion `px != 0' failed.
```

### Steps to reproduce the behavior

* start MantidPlot
* set facility to ISIS
* load live data
* select instrument=MARI
* select connection=kafka_event
* click run
* wait until `MonitorLiveData` has started and click cancel => :boom: 

### Platforms affected

Verified on Ubuntu 16.04 but I would assume all.

mducle, 2019-02-18:
@martyngigg Thanks for spotting this - it might be the reason behind this issue: #22939

TTitcombe, 2019-03-12:
Unable to replicate. Tested on Ubuntu 18.04. Possible that issue arises when beam is off.

closed 2021-03-02 by stale[bot]