mantidproject/mantid #24819: Workbench: ordered_recovery doesn't format path properly for Windows

DTasev, 2019-02-19:
```python
from mantid.simpleapi import *

Load(Filename=r'C:\Users\qbr77747\dev\2m\source\build\ExternalData\Testing\Data\UnitTest\SingleCrystalPeakTable.nxs', OutputWorkspace='peek') # 2019-02-19T09:27:01.674971000
Load(Filename=r'C:\Users\qbr77747\dev\2m\source\build\ExternalData\Testing\Data\UnitTest\SavedTableWorkspace.nxs', OutputWorkspace='table') # 2019-02-19T09:27:04.095955000
Load(Filename=r'C:\Users\qbr77747\dev\2m\source\build\ExternalData\Testing\Data\UnitTest\irs26173_graphite002_red.nxs', OutputWorkspace='matrix') # 2019-02-19T09:27:04.189943000
```

martyngigg, 2019-02-19:
Could we update the issue with some of the error output, please?

martyngigg, 2019-02-19:
So we are basically see a problem with the fact that we are now using `unicode_literals` and `\u` is interpreted as a unicode escape sequence. `\Users` is treated as an invalid code point and we get the syntax error.

PR #24834 (2019-02-20): Prevent __future__ imports being inherited by the CodeEditor execute environment
**Description of work.**
Adds the `dont_inherit` flag to `compile` which specifies to _not_ inherit any `__future__` imports from the calling environment.

This fixes an issue preventing ProjectRecovery to execute the generated `ordered_recovery.py`.

**To test:**
- Run `print(type("\ua"))`.
- It should show error: "SyntaxError: (unicode error) 'unicodeescape' codec can't decode bytes in position 0-2: truncated \uXXXX escape"
- Remove `unicode_literals` from `__future__` import
- Run again, it should run without a [... cut]
closed 2019-02-20 by peterfpeterson