mantidproject/mantid #24849: Workbench: comment the line where the cursor is when no selection

DTasev, 2019-02-21:
### Expected behavior
Pressing `Ctrl+/` comment-line shortcut should comment out the line on which the cursor is currently, if there is no selection.

### Actual behavior
It does nothing, to comment out the line you need to select at least 1 character and then press `Ctrl+/`.

### Steps to reproduce the behavior
Move the cursor on any line, press `Ctrl+/`.

### Platforms affected

hsautessella, 2019-03-28:
Do you think it would be worth refactoring the code that performs the commenting into its own class? This would keep the `PythonFileInterpreter` class shorter and make it easier to write some tests for the commenting functionality.

DTasev, 2019-03-28:
Yeah, I think it would benefit to have it in a separate class

PR #25424 (2019-04-01): Workbench: Enable comment current line in script editor with no selection
**Description of work.**
When there is nothing selected in the script editor toggle comment will now comment/uncomment the line the cursor is on.

The code that carries out the commenting/uncommenting has been refactored into its own class and some unit tests have been written for that class.  This cuts down the size of the `PythonFileInterpreter` class and makes testing the commenting functionality easier.

**To test:**
- Run tests in `test_codecommenter.py`.
- Open workbench and check that commenting code works as expe [... cut]
closed 2019-04-01 by peterfpeterson