mantidproject/mantid #24869: AppendSpectra can create spectrum with number of 0

SamJenkins1, 2019-02-22:
if two workspaces containing only one spectra are appended, the output workspace can have two spectra whose workspace index and spectra are both 0 and 1. This could cause issues in system tests as saving/loading the file appears to correct the spectrum number to 1 and 2, if these are used as a reference file with compare workspace, the spectrum numbers will not match.

soininen, 2019-02-25:
Please note that the spectrum numbers are not always stored in the NeXus files. PR #24708 tries to fix this problem.

SamJenkins1, 2019-02-25:
That should fix the problem with tests so long as new reference files are made, but surely still appending tow workspaces with spectrum number 1 should create 1 and 2 not 0 and 1? spectrum numbers are supposed to start at 1 right?

soininen, 2019-02-25:
For fresh workspaces the spectrum numbering starting from 1 is generally true but not always. For example, `LoadILLReflectometry` gives spectrum numbers starting from 0 because we felt that there should be 1-to-1 correspondence between the spectrum numbers and workspace indices due to visualization reasons.

`AppendSpectra` should preserve the original spectrum numbers as long as there are no conflicts. So when appending spectra with spectrum numbers 8 and 13 you should get a workspace with those numbers. Maybe if one tries to append two spectra with the same spectrum numbers the output workspace should renumber the spectra starting from 1.

closed 2020-07-29 by gemmaguest