mantidproject/mantid #24939: Unscripted Testing ISIS Indirect interfaces

NickDraper, 2019-03-04:
1. Read http://www.mantidproject.org/Unscripted_Manual_Testing
1. Comment against this ticket the OS environment you are testing against.
1. Don't spend more than a few hours on the testing as fatigue will kick in.
1. If you find errors in the possibly scant documentation, then correct them.
1. Close the this issue once you are done.

If you find bugs:
* Search to see if an issue already exists
* create an issue if it is new
 * Urgent bugs or crashes should by against the current release, and assigned to a developer, then go and talk to the developer if possible.
 * Less urgent bugs should be against a subsequent release, and assigned if the correct developer is known.

### Specific Notes:

Follow the directions at: http://developer.mantidproject.org/Testing/IndirectInelastic/IndirectInelasticAcceptanceTests.html

AnthonyLim23, 2019-03-04:
The reduction tab jumps back to the top of the GUI when run is pressed. - annoying but not breaking anything.

For the second test at step 10 I end up with 51 spectra not 50 

In the last part of the second test the contour plot seems to be empty (all zeros)

For the I(Q,T) I got the error 
```Error in execution of algorithm TransformToIqt:
float division by zero
  at line 81 in 'C:/Users/BTR75544/nightly/MantidNightlyInstall/plugins/python/algorithms\WorkflowAlgorithms\TransformToIqt.py'
  caused by line 162 in 'C:/Users/BTR75544/nightly/MantidNightlyInstall/plugins/python/algorithms\WorkflowAlgorithms\TransformToIqt.py'```
using  26184 to 5 with both  sample and resolution selected
![image](https://user-images.githubusercontent.com/25006392/53743040-e08c9380-3e91-11e9-9434-85153240a218.png)

AnthonyLim23, 2019-03-04:
@rob could you have a look at these and let us know if they are important? If they are I will make issues.

closed 2019-03-08 by AnthonyLim23