mantidproject/mantid #24941: Unscripted Testing SCD event data reduction

NickDraper, 2019-03-04:
1. Read http://www.mantidproject.org/Unscripted_Manual_Testing
1. Comment against this ticket the OS environment you are testing against.
1. Don't spend more than a few hours on the testing as fatigue will kick in.
1. If you find errors in the possibly scant documentation, then correct them.
1. Close the this issue once you are done.

If you find bugs:
* Search to see if an issue already exists
* create an issue if it is new
 * Urgent bugs or crashes should by against the current release, and assigned to a developer, then go and talk to the developer if possible.
 * Less urgent bugs should be against a subsequent release, and assigned if the correct developer is known.

VickieLynch, 2019-03-06:
Only found problem with rarely used Pick Mode for VSI only displaying information about first point selected. Will create issue for future release.

closed 2019-03-06 by VickieLynch

VickieLynch, 2019-03-06:
Tested on Rhel7