mantidproject/mantid #24943: Unscripted Testing DGSReduction

NickDraper, 2019-03-04:
1. Read http://www.mantidproject.org/Unscripted_Manual_Testing
1. Comment against this ticket the OS environment you are testing against.
1. Don't spend more than a few hours on the testing as fatigue will kick in.
1. If you find errors in the possibly scant documentation, then correct them.
1. Close the this issue once you are done.

If you find bugs:
* Search to see if an issue already exists
* create an issue if it is new
 * Urgent bugs or crashes should by against the current release, and assigned to a developer, then go and talk to the developer if possible.
 * Less urgent bugs should be against a subsequent release, and assigned if the correct developer is known.

### Specific Notes:

Follow:  http://www.mantidproject.org/Direct:DgsReduction
     and http://www.mantidproject.org/Create_MD_Workspace_GUI#Direct_Inelastic_Mod_Q

AnthonyLim23, 2019-03-15:
it opens

NickDraper, 2019-03-20:
The time for fi=urther testing on this has run out.  Closing for this release.

closed 2019-03-20 by NickDraper