mantidproject/mantid #24948: Unscripted Testing Muon

NickDraper, 2019-03-04:
1. Read http://www.mantidproject.org/Unscripted_Manual_Testing
1. Comment against this ticket the OS environment you are testing against.
1. Don't spend more than a few hours on the testing as fatigue will kick in.
1. If you find errors in the possibly scant documentation, then correct them.
1. Close the this issue once you are done.

If you find bugs:
* Search to see if an issue already exists
* create an issue if it is new
 * Urgent bugs or crashes should by against the current release, and assigned to a developer, then go and talk to the developer if possible.
 * Less urgent bugs should be against a subsequent release, and assigned if the correct developer is known.

### Specific Notes:

See instructions at http://docs.mantidproject.org/v3.9.0/interfaces/Muon_Analysis_unscripted_testing_guide.html
http://developer.mantidproject.org/Testing/MuonInterface/MuonTesting.html

Mantid-Matthew, 2019-03-07:
I am testing the ACL interface on Ubuntu 16.04. I've run into the following issues:
 * The load button will not do anything if you have entered some invalid values with no indication of what is wrong. I.e. try selecting custom grouping and then not entering anything in the boxes
 * The GUI doesn't check that the default instrument is actually a Muon one
 * If you try to load a non-existent file or try to click the load button before the archive search is complete you will get an error message saying you have not specified a file. A more helpful error message would be to state that the file you have specified cannot be found.
 * For multiperiod data if you export results and then try and import results you will get an error message complaining that the loaded data has more than one spectrum
 * If you enter some invalid values in the Grouping boxes such as negative numbers you will get an uncaught exception, the GUI will then remain permanently disabled.

alicerussell1, 2019-03-08:
Testing the muon analysis interface on Windows. I've found the following:

- the co-add/simultaneous button in the data analysis tab is not disabled on an individual fit
- when you load the current run for EMU then click the left arrow and the right arrow (i.e. return to the current run) it cannot find it. Does not find it when searched for either. Only happens for EMU instrument

Testing the new frequency domain interface on Windows I found the following issues:

- If you input two negative values into the group asymmetry range in grouping tab when data is loaded, this causes Mantid to close. Only happens if the numbers are in the correct order if the first number is bigger you get the expected error message.
- In the grouping tab if you change detector ID to a 0 it deletes the group - I would expect it to just return an error
- In the MaxEnt transform when 0 is entered as the lagrange multiplier for chi^2 it return an uncaught exception
- Uncaught error when 0 is used for inner iterations
- Uncaught error when 0 is used fir last good time

The last three errors also apply to the old frequency domain interface.

NickDraper, 2019-03-08:
@AnthonyLim23 Do you want to have a look at these and see which we may want to address?

AnthonyLim23, 2019-03-08:
I think we should fix:

1. the hard crash in ALC
2. @Mantid-Matthew  could you look at the first two for the new GUI?
3. The last three just need a validation

AnthonyLim23, 2019-03-15:
is all of the testing done?

closed 2019-03-19 by AnthonyLim23