mantidproject/mantid #24950: Unscripted Testing ISIS SANS

NickDraper, 2019-03-04:
1. Read http://www.mantidproject.org/Unscripted_Manual_Testing
1. Comment against this ticket the OS environment you are testing against.
1. Don't spend more than a few hours on the testing as fatigue will kick in.
1. If you find errors in the possibly scant documentation, then correct them.
1. Close the this issue once you are done.

If you find bugs:
* Search to see if an issue already exists
* create an issue if it is new
 * Urgent bugs or crashes should by against the current release, and assigned to a developer, then go and talk to the developer if possible.
 * Less urgent bugs should be against a subsequent release, and assigned if the correct developer is known.

### Specific Notes:

Follow: http://www.mantidproject.org/SANS_Data_Analysis_at_ISIS

aybamidele, 2019-03-04:
This was done on Windows

aybamidele, 2019-03-05:
I found some bugs and have created issues for them.

closed 2019-03-05 by aybamidele