mantidproject/mantid #24953: Unscripted Testing ORNL SANS

NickDraper, 2019-03-04:
1. Read http://www.mantidproject.org/Unscripted_Manual_Testing
1. Comment against this ticket the OS environment you are testing against.
1. Don't spend more than a few hours on the testing as fatigue will kick in.
1. If you find errors in the possibly scant documentation, then correct them.
1. Close the this issue once you are done.

If you find bugs:
* Search to see if an issue already exists
* create an issue if it is new
 * Urgent bugs or crashes should by against the current release, and assigned to a developer, then go and talk to the developer if possible.
 * Less urgent bugs should be against a subsequent release, and assigned if the correct developer is known.

ricleal, 2019-03-11:
Tested the regular HFIR and EQSANS reduction algorithms. All working.

ricleal, 2019-03-11:
Not working!!!

```
Error in execution of algorithm SANSReduction
Cannot create algorithm, invalid string format.
  at line 31 in '/opt/mantidnightly/plugins/python/algorithms/WorkflowAlgorithms/SANSReduction.py'
  caused by line 255 in '/opt/mantidnightly/plugins/python/algorithms/WorkflowAlgorithms/SANSReduction.py'

```

ricleal, 2019-03-12:
@martyngigg It looks like your changes: PR #24572 break ORNL SANS Workflow.

I did some tests and if you set properties (int/string) in a C++ algorithm to a python algorithm (C++ everything it's OK) they appear null.

I have debugged the code but find nothing. I just know it works before your changes but not after. https://github.com/mantidproject/mantid/blob/master/Framework/WorkflowAlgorithms/src/SetupEQSANSReduction.cpp#L864

You can see with this bit of code:

```python
import mantid
from mantid.simpleapi import *
from reduction_workflow.instruments.sans.sns_command_interface import *

def print_properties(pm):
    for k,v in zip(pm.keys(), pm.values()):
        print("{} :".format(k), end="")
        try:
            print(v.value)
        except:
            print(v)

# properties are null when the default is changed

# Defaults
PropertyManagerDataService.clear()
SetupEQSANSReduction()
pm = PropertyManagerDataService.retrieve('__sans_reduction_properties')
print_properties(pm)
#IQAlgorithm :{"name":"EQSANSAzimuthalAverage1D","properties":{"ComputeResolution":false,"OutputWorkspace":"__TMP0x7f44e40e14c0"},"version":1}

# 1. Let's change a property that will be a int/string input to python algorithm
SetupEQSANSReduction(IQNumberOfBins=101)
pm = PropertyManagerDataService.retrieve('__sans_reduction_properties')
# See IQAlgorithm: None
print_properties(pm)
# Output:
# IQAlgorithm :None

# 2. Let's change a property that will be a int/string input to C++ algorithm
SetupEQSANSReduction(IQ2DNumberOfBins=101)
pm = PropertyManagerDataService.retrieve('__sans_reduction_properties')
# See IQXYAlgorithm: OK changed NumberOfBins
print_properties(pm)
# Output:
# IQXYAlgorithm :{"name":"EQSANSQ2D","properties":{"NumberOfBins":101},"version":1}
```

martyngigg, 2019-03-13:
@ricleal Sorry about that. I had hoped that the good system test coverage for ORNL SANS was enough to confirm those changes were okay. I will look into it.

martyngigg, 2019-03-18:
@ricleal Is there anything further to be tested here or is this okay now after the fix in #25138?

ricleal, 2019-03-18:
@martyngigg All OK! I tested it over the weekend. Thanks.

martyngigg, 2019-03-18:
That's great. Thanks!

closed 2019-03-18 by martyngigg