mantidproject/mantid #24954: Unscripted Testing Multi dataset fitting

NickDraper, 2019-03-04:
1. Read http://www.mantidproject.org/Unscripted_Manual_Testing
1. Comment against this ticket the OS environment you are testing against.
1. Don't spend more than a few hours on the testing as fatigue will kick in.
1. If you find errors in the possibly scant documentation, then correct them.
1. Close the this issue once you are done.

If you find bugs:
* Search to see if an issue already exists
* create an issue if it is new
 * Urgent bugs or crashes should by against the current release, and assigned to a developer, then go and talk to the developer if possible.
 * Less urgent bugs should be against a subsequent release, and assigned if the correct developer is known.

### Specific Notes:

Follow: https://www.mantidproject.org/Fitting_QENS_I(Q,_t)

DTasev, 2019-03-05:
Testing this on Windows

mantid-roman, 2019-03-05:
Yes, I can fix it.

PR #25232 (2019-03-20): Fix multi dataset fitting bugs
**Description of work.**
Always use the single-dataset fit if only 1 spectrum is fitted.
Remove the crystal field functions from gui dialogs.
Improved handling of workspace groups.

**To test:**
1. Try sequential and simultaneous fitting of a single spectrum. The result should be the same.
2. Check that there are no CrystalField* functions in the Add Function dialog.
3. Check that workspace groups that don't contain matrix workspaces don't show in the Add Workspace dialog.

Fixes #24954

---

#### Reviewer ####

Please comment on the following ([full description] [... cut]
closed 2019-03-20 by NickDraper