mantidproject/mantid #24957: Unscripted Testing Vates

NickDraper, 2019-03-04:
1. Read http://www.mantidproject.org/Unscripted_Manual_Testing
1. Comment against this ticket the OS environment you are testing against.
1. Don't spend more than a few hours on the testing as fatigue will kick in.
1. If you find errors in the possibly scant documentation, then correct them.
1. Close the this issue once you are done.

If you find bugs:
* Search to see if an issue already exists
* create an issue if it is new
 * Urgent bugs or crashes should by against the current release, and assigned to a developer, then go and talk to the developer if possible.
 * Less urgent bugs should be against a subsequent release, and assigned if the correct developer is known.

### Specific Notes:

# VSI Testing

<div class="contents">

</div>

*Preparation*

  - Get the scripts from the Mantid repository [this
    directory](https://github.com/mantidproject/mantid/tree/master/scripts/Vates).
  - Make sure that you have the Mantid system test data. You can do this
    by building a SystemTestData target of Mantid.
  - Put the system test data directory in your Mantid user directories

**Time required 20-30 minutes**

-----

1.  Load the script SXD\_NaCl.py in MantidPlot. Run it.
2.  This should load a workspace called QLab. Right-click this workspace
    and choose 'Show Vates simple interface'.
3.  This should open the VSI window. Like the picture below. Check that
    all tabs are present.

[image]

1.  Click the various tabs making sure that the interface does not
    crash.
2.  Make sure all tabs have a tool tip when you hover over them.
3.  Go into three slice mode (the star shaped tab at the top).
4.  Play with the maximum and minimum parameters at the top of the plot,
    make sure the colours change in the plot.
5.  Hover on each of the upper two plots and the lower left plot. While
    hovering over a plot use the roll bar (or equivalent), the slices in
    the other plots should update. The lower right plot should show all
    three slices. Check that the text on the plots is legible and
    updates appropriately.
6.  Switch to multi-slice view (just to the left of three slice view).
    1.  In the properties tab (left of the plot) select the Display
        menu.
    2.  Choose all options for Representation, make sure they work.
    3.  Change the Styling and Lighting sliders, make sure the picture
        updates.
    4.  Next to Data Axes Grid click edit, this should open a new dialog
        of settings for the plot.
    5.  Change some of the properties in this dialog and click apply,
        the plot should update accordingly.
    6.  Make sure that the sliders for the plot work. These are the
        triangles on the scales above and to the left of the plot. Slide
        them and move through the plot.
    7.  On the slider scale Shift+Right-click should add another slice
    8.  Select the View menu.
    9.  Turn on/off the orientation axes visibility, check that the
        little axes icon on the plot turns on and off.
    10. Check and uncheck the parallel camera box, make sure the view
        switches between perspective on and off.
7.  Switch to standard view (left of multi-slice)
    1.  In the Rebin drop-down (above the plot) select Fast Rebin and
        click Slice - a new MDScaleWorkspace should appear in the list
    2.  In the Properties tab select Properties and change the X/Y/Z
        scaling factors, click Apply. The plot should distort
        accordingly.
    3.  In the properties tab click Delete and the plot alterations
        should be removed from the plot.
    4.  Repeat the previous three steps for the different Rebin options.
8.  Close the VSI interface and then reopen as before. You are now in
    scatter plot mode.
    1.  Under Properties, change the Number of Points and Top Percentile
        settings and see that the plot changes.
    2.  Drag the peaks\_qLab workspace from the main GUI onto VSI. This
        should show the peak position in the plot as points.
    3.  Drag the peaks\_qLab\_integrated workspace from the main GUI
        onto VSI. This should show the peak positions in the plot as
        spheres.
9.  Close the VSI GUI.

gemmaguest, 2019-03-04:
@OwenArnold Nick said you didn't have any unscripted testing yet, so would you be able to pick up this one?

OwenArnold, 2019-03-06:
One issue found and logged

closed 2019-03-06 by OwenArnold