mantidproject/mantid #24958: Unscripted Testing Engineering Diffraction

NickDraper, 2019-03-04:
1. Read http://www.mantidproject.org/Unscripted_Manual_Testing
1. Comment against this ticket the OS environment you are testing against.
1. Don't spend more than a few hours on the testing as fatigue will kick in.
1. If you find errors in the possibly scant documentation, then correct them.
1. Close the this issue once you are done.

If you find bugs:
* Search to see if an issue already exists
* create an issue if it is new
 * Urgent bugs or crashes should by against the current release, and assigned to a developer, then go and talk to the developer if possible.
 * Less urgent bugs should be against a subsequent release, and assigned if the correct developer is known.

### Specific Notes:

Follow: http://docs.mantidproject.org/nightly/interfaces/Engineering%20Diffraction%20Test%20Guide.html

AnthonyLim23, 2019-03-04:
I will double check, but I may need to swap (if it is still bricked from last time)

AnthonyLim23, 2019-03-04:
I still cannot test this due to the bug in #24242, I suspect the same will be true for @Pasarus

AnthonyLim23, 2019-03-05:
On the nightly using https://isis.analysis.stfc.ac.uk:

Open the GUI and put 1 for RB number
Type the run numbers given in the link above into the second set of boxes
Click calibrate
Hard crash

AnthonyLim23, 2019-03-08:
If I load the data via browse it works. I will do the testing on Tuesday

AnthonyLim23, 2019-03-13:
1. When I load the data, the vanadium curve plot is not automatically produced
2. The testing doc has suggestions that completely broke my copy of the interface 
3. The focus run needs to be browsed to (this may be because I have previously broken the interface)
4. Using 236516 in focus caused my computer to freeze

AnthonyLim23, 2019-03-13:
5. the instructions for pr-processing are insufficient as I do not know which data to use
6. I was in fitting and loaded the data for 241391 bank 1. I clicked select peak and clicked one at about 49500. Then I clicked fit - hard crash

AnthonyLim23, 2019-03-13:
I dont have the file for the GSAS fitting

SamJenkins1, 2019-03-13:
1. The plotting on the calibration tab is only run on newly generated files so the fitting can be checked, if the file has already been generated it should have already been check and not need this process.
2. I'll make an issue for this.
3. This appears to be due to the broken interface
4. Same as above
5. should I add the files I sent to youas suggestion to test files in the documentation? 
6. From a brief look into this it looks like it might be due to a similar problem that was causing the calibration crash (gui stuff happening in the wrong thread, Ill look into it further)
7. This is listed in the Peak Tools `How to use` on the gui
8. I believe this may be caused by the same root problem as 6.

AnthonyLim23, 2019-03-13:
5. yes please
7. Is another name possible as it is a little confusing. I would assume `add peak` would add the peak.

SamJenkins1, 2019-03-13:
What would you suggest renaming it to? it does add the peak to the list of peaks to attempt to fit

AnthonyLim23, 2019-03-13:
I have had another play with it. It might just be better to disable the add peak button as long as there is no peak selector on the plot. What do you think?

SamJenkins1, 2019-03-13:
I'll have a look into it,

closed 2019-03-15 by AnthonyLim23