mantidproject/mantid #24968: Unscripted testing - project recovery

NickDraper, 2019-03-04:
1. Read http://www.mantidproject.org/Unscripted_Manual_Testing
1. Comment against this ticket the OS environment you are testing against.
1. Don't spend more than a few hours on the testing as fatigue will kick in.
1. If you find errors in the possibly scant documentation, then correct them.
1. Close the this issue once you are done.

If you find bugs:
* Search to see if an issue already exists
* create an issue if it is new
 * Urgent bugs or crashes should by against the current release, and assigned to a developer, then go and talk to the developer if possible.
 * Less urgent bugs should be against a subsequent release, and assigned if the correct developer is known.

### Specific Notes:

http://developer.mantidproject.org/Testing/ErrorReporter-ProjectRecovery/ProjectRecoveryTesting.html

gemmaguest, 2019-03-05:
All seems ok on Ubuntu 16.04 apart from a known issue where if you re-open MantidPlot quickly after a crash the recovery dialog is not shown. Also I think the instructions in test 5 are a bit out of date: the error dialog is not shown; and the workspace name should be `Instance 1` rather than `NewWorkspace1`

gemmaguest, 2019-03-05:
I get an unexpected exception `Couldn't open project file` when recovering from the second script in 
test 1 on Windows 10.

NickDraper, 2019-03-20:
Specific issue is being tracked in the project, so I'll close this issue

closed 2019-03-20 by NickDraper