mantidproject/mantid #24969: Unscripted testing - error reporter

NickDraper, 2019-03-04:
1. Read http://www.mantidproject.org/Unscripted_Manual_Testing
1. Comment against this ticket the OS environment you are testing against.
1. Don't spend more than a few hours on the testing as fatigue will kick in.
1. If you find errors in the possibly scant documentation, then correct them.
1. Close the this issue once you are done.

If you find bugs:
* Search to see if an issue already exists
* create an issue if it is new
 * Urgent bugs or crashes should by against the current release, and assigned to a developer, then go and talk to the developer if possible.
 * Less urgent bugs should be against a subsequent release, and assigned if the correct developer is known.

### Specific Notes:

http://developer.mantidproject.org/Testing/ErrorReporter-ProjectRecovery/ErrorReporterTesting.html

gemmaguest, 2019-03-05:
The error reporter is not showing on Windows 10

gemmaguest, 2019-03-06:
Looks good in MantidPlot in Windows 10 following the fix in #25029. 

Viewing the sample logs as per the instructions no longer produces an unexpected exception. I used the test in #23114 to produce one instead.

gemmaguest, 2019-03-06:
I'm not sure how to produce an unexpected exception in Workbench

gemmaguest, 2019-03-13:
I did the MantidPlot testing at 5pm on 6th March. There should be Public, Public2 and Public3 named entries reported and one non-identifiable information shared. No name/email entries containing ‘Private’ should have been shared.

I did the workbench tests at 4.30pm on 13th March. There should be the same entries as above for this date. Assuming those reports were received all looks good :+1:

Note that currently you can produce an unexpected exception in Workbench by opening the SANS GUI, going to the Diagnostics tab and clicking `Horizontal Integral`.

gemmaguest, 2019-03-14:
Database entries look fine

closed 2019-03-14 by gemmaguest