mantidproject/mantid #24972: Unscripted Testing Mantid Group Workbench

Pasarus, 2019-03-04:
1. Read http://www.mantidproject.org/Unscripted_Manual_Testing
1. Comment against this ticket the OS environment you are testing against.
1. Don't spend more than a few hours on the testing as fatigue will kick in.
1. If you find errors in the possibly scant documentation, then correct them.
1. Close the this issue once you are done.

If you find bugs:
* Search to see if an issue already exists
* create an issue if it is new
 * Urgent bugs or crashes should by against the current release, and assigned to a developer, then go and talk to the developer if possible.
 * Less urgent bugs should be against a subsequent release, and assigned if the correct developer is known.

### Specific Notes:

See https://www.mantidproject.org/Unscripted_Manual_Testing#Groups

ElliotAOram, 2019-03-06:
Windows 10

ElliotAOram, 2019-03-06:
**Qt help Documentation**: Not currently implemented
**Data Loading**: OK
**Load/Save Project**: 
* No progress bar when saving the project - very confusing what (if anything is happening during this wait period
* See above but for opening project
* Can we do something better with the structure of the project save. For example project save save a `MTDPROJ` and all `.nxs` files but these are just put alongside the file so if you just save the project your documents folder then you end up with a bunch of nxs files in there. IMO a better option would be to add all this into a folder with the project name (this is then more obvious that the nxs files relate to the saved project)
* HIGH: Can force a mantid crash if you do the following:
  * open existing project
  * save project as...
  * Select the existing project
  * Click yes to replace it 
  * see that workbench crashes

**1D graphs**: OK
**2D Graphs**: OK
**3D Graphs**: Not available in workbench
**Matplotlib**: OK
**Table Workspace display**: OK
**Matrix workspace display**: OK
**Printing**:
  * Bug:
    * plot a spectrum
    * right click the print icon
    * left click the empty item in the drop down menu
    * Observe that the plot menu disappears and you can't get it back

Pasarus, 2019-03-06:
> **Load/Save Project**:
>
>     * No progress bar when saving the project - very confusing what (if anything is happening during this wait period
>
>     * See above but for opening project
>
>     * Can we [...]

Since I wrote Project Save/Load I shall reply here.

- With regards to first point: You make a valid arguement and I agree however at present that requires a GUI being made and that is not probable for this release cycle, however is planned for the future.
- Second point: See first
- Third Point: I agree it would be better to do that, (it was actually the first implementation of project save/load), however the current behavior is inline with what our current MantidPlot windows users experience.
- Fourth Point: That is a bug in the code that is likely an oversight from where the change in behavior occurred, I will work on that now in a high priority issue.

martyngigg, 2019-03-06:
*HIGH: Can force a mantid crash if you do the following ...*. Fixed in #25043

ElliotAOram, 2019-03-07:
All bugs etc. raised from this unscripted testing are being dealt with in their own issues - as such this issue can be closed

closed 2019-03-07 by ElliotAOram

hsautessella, 2019-03-11:
Sorry I managed to reference the wrong issue here. Please ignore