mantidproject/mantid #24983: Indirect unscripted testing bugs

AnthonyLim23, 2019-03-04:
Let me know if some of these are not worth fixing. 

The reduction tab jumps back to the top of the GUI when run is pressed. - annoying but not breaking anything.

For the second test at step 10 I end up with 51 spectra not 50

In the last part of the second test the contour plot seems to be empty (all zeros)

For the I(Q,T) I got the error

float division by zero
  at line 81 in 'C:/Users/BTR75544/nightly/MantidNightlyInstall/plugins/python/algorithms\WorkflowAlgorithms\TransformToIqt.py'
  caused by line 162 in 'C:/Users/BTR75544/nightly/MantidNightlyInstall/plugins/python/algorithms\WorkflowAlgorithms\TransformToIqt.py'```
using  26184 to 5 with both  sample and resolution selected
[image]

robertapplin, 2019-03-04:
The test should say 51 spectra rather than 50 - this is the case in the Mantid version 3.13.
The other two issues couldn't be reproduced by me or @AnthonyLim23

SpencerHowells, 2019-03-04:
@AnthonyLim23 @robertapplin For I(q,t) it looks as if the 2 data sets are incompatible. 1 is graphite002 & the other is graphite004. It shouldn’t allow that.

PR #24995 (2019-03-08): Indirect Unscripted testing changes
**Description of work.**
This PR updates some of the acceptance testing steps. It also adds a check for analysers and reflections for the Iqt interface before performing TransformToIqt.

**To test:**
1.`Interfaces`->`Indirect`->`Data Analysis`->`Iqt` interface
2. Load the data below and click `Run`. An error message stating the reflections are different should appear.

Check the changes to the acceptance tests.

*No release notes required as the bug is relatively minor and was probably unnoticed*

Fixes #24983

#### Reviewer ####

Please comment on the follo [... cut]
closed 2019-03-08 by rosswhitfield