mantidproject/mantid #24992: Enginex hard crash

AnthonyLim23, 2019-03-05:
On the nightly using https://isis.analysis.stfc.ac.uk:

Open the GUI and put 1 for RB number
Type the run numbers given in the link above into the second set of boxes
Click calibrate
Hard crash

SamJenkins1, 2019-03-05:
The link does not appear to contain run-numbers to use?

AnthonyLim23, 2019-03-05:
```Vanadium#: ENGINX00236516.nxs
Calibration#: ENGINX00241391.nxs```

SamJenkins1, 2019-03-05:
what OS were you running on? on my linux machine those inputs run fine, and my windows vm simply fails because its not connected to the data archive.

AnthonyLim23, 2019-03-05:
I did it on https://isis.analysis.stfc.ac.uk/#/login which is a linux

SamJenkins1, 2019-03-06:
Which environment did you use Anthony? I can't reproduce seem to reproduce the crash at least on the excitations environment.

AnthonyLim23, 2019-03-06:
I used Excitation's 8 CPU 32GB

SamJenkins1, 2019-03-06:
When you ran through it with me it crashed when you got the files from browsing to them, right?

AnthonyLim23, 2019-03-06:
yep

SamJenkins1, 2019-03-06:
never mind, I've began to be able to reproduce it on an analysis environment... without having changed what inputs I was using... or anything really... I am now extremely confused as to what could possibly have changed

SamJenkins1, 2019-03-06:
think I've worked out what causes it at least, looks like its what the calibration folder is set to.

AnthonyLim23, 2019-03-06:
how do I fix it?

SamJenkins1, 2019-03-07:
you should be able to get around this crash by going to settings tab and setting the calibration folder to any folder it can write to, I'm looking into fixing the crash now, it looks like its caused when it tries to open the warning caused by having no calibration folder set

SamJenkins1, 2019-03-07:
looks like this was actually also present in the previous release

PR #25058 (2019-03-11): Fix crash upon running enginX's gui calibrate with no calibration directory selected
**Description of work.**
The crash occurred due to  trying to call the message-box stating that a calibration folder was not set, not being called on the gui thread, changing to `invokeMethod` lead to the crash no longer happening, but the message-box also not appearing, to fix the crash the message-box has been replaced with a warning in the logger.

**To test:**
Check that the settings tab on the enginX gui does not have any path set in `Calibration settings->Input directories/folder:->calibration`
go back t [... cut]
closed 2019-03-11 by rosswhitfield