mantidproject/mantid #24997: Contour plot with text axis

AnthonyLim23, 2019-03-05:
Bug from James. 

The sequence of events was:
-	Load some runs (probably need 3 or more). Optionally pre-process with AsymmetryCalc, FFT or whatever.
-	Do ConjoinSpectra on the set of runs, with “LabelUsing” being some value that varies in order from run to run (run number would do).
-	Right click the joined workspace and do “Color Fill Plot”
You get the sad Mantid error reporting dialog flickering on and off. Selecting it from the task bar makes it stay active. The choices here are “Continue” which returns to the flickering dialog or “Terminate” which of course kills Mantid.

I have a modified version “ConjoinSpectraNumAx” (attached) which makes the spectrum axis numeric and it does work with Color Fill Plot. The standard one creates a text axis.

The bug is present in the latest nightly build and 3.13. Previously (a while since I did this) it would just give a one-off error message and I could continue working.

It’s easier to run ConjoinSpectra from a script, rather than having to type a list of workspace names into the dialog, especially if they’re the long ones from Muon Analysis! When ADSValidator’s Python API is fixed (see my recent Forum report), this is one algorithm that could benefit.

Pasarus, 2019-03-14:
Add safety check to plotting and fix conjoined spectra, being strings.

Pasarus, 2019-03-19:
This issue will be completed after the release by @DTasev because he has made the majority of changes to the logic in ConjoinSpectra. 

Whilst the part of this issue where MantidPlot crashed will be fixed before release.

closed 2019-04-29 by AnthonyLim23