mantidproject/mantid #25001: Offline help window navigation buttons do not update correctly

hsautessella, 2019-03-05:
**Original reporter:** hsautessella. 

### Expected behavior
- The back button should be enabled whenever a forward navigation is made within the window.
- The forward button should only be enabled after the back button is pressed.

### Actual behavior
The back button is not enabled after the first forward navigation inside the help window. You must make another forward navigation before the back button is enabled.

The forward button is not always disabled after a forward navigation is made.

### Steps to reproduce the behavior
- Launch MantidPlot and open the offline help: `Help`->`MantidPlot Help`.
- Click the `Mantid Basic Course` link.
[image]
- The back button will not have been enabled
[image]
- Follow the `Setting Up and Getting Started` link. The back button will now be enabled and will allow you to go back the two pages you have navigated. Do this.
- Once back at the homepage follow the `Mantid Basic Course` link again. You should see the back button is not enabled, but the forward button is.
[image]

### Platforms affected
Tested on Windows 10 with Nightly v3.13.20190301.1751

NickDraper, 2019-03-05:
This is interesting, but not something we have any control over.  The QtAssistant app is a black box to us on this.  I propose we don't attempt to fix this.

closed 2019-03-05 by NickDraper

hsautessella, 2019-03-19:
I know how to fix this if you think it's worth doing? It may be since this is likely to be ported to the Workbench later. 

The issue is that the `updateNavButtons()` function checks the history to see if it can go forward or backward, however it's called before the history is updated, so it's always a step behind.