mantidproject/mantid #25004: Do not open same file in multiple tabs

peterfpeterson, 2019-03-05:
When running `mantidworkbench -x <filename>`, currently the workbench will open it in a new tab. This can leave a user with several tabs all looking at the same file.

Pasarus, 2019-03-05:
This is actually possible in MantidPlot if you just continually open them in the script window. What is interesting is if you restart MantidPlot only one of them will open back up.

I brought this up with @hsautessella and I think it is intended?

hsautessella, 2019-03-05:
When starting workbench it will restore whatever tabs were open when it was last closed. So if a user had multiple instances of the same file open, both instances will be restored.

Is the wanted behaviour to not allow multiple instances of the same file (as Sam said, this is possible on MantidPlot), or to not allow multiple instances on startup, or to not allow an extra instance to be created via the `-x` flag if the same file is being restored?

peterfpeterson, 2019-03-05:
On problem was that workbench wasn't (a few days ago) allowing to save changes to the file if it was open in multiple tabs. I didn't like the behavior as I frequently (for development/testing) create a script and `mantidworkbench -x script` then poke about. I had 10+ tabs open with the same file before I first noticed it.

hsautessella, 2019-03-05:
It should be straightforward to have it so that multiple instances of the same tab won't be restored, mimicking the MantidPlot behaviour, if you'd like me to do that?

peterfpeterson, 2019-03-05:
Since it is in the same area. Can you also print a warning (rather than error) if a file that was open last time no longer exists?

hsautessella, 2019-03-06:
I've created a PR for the issue, but as far as I can tell we already print a warning and not an error: 
https://github.com/mantidproject/mantid/blob/26f4f3052304176d0c2bda0d7d8cf2bfe02866a7/qt/applications/workbench/workbench/plugins/editor.py#L132-L134

PR #25028 (2019-03-06): Remove duplicate tabs in session restore
**Description of work.**
Previous behaviour was to restore whatever tabs the user had open the last time workbench was closed. If a user had multiple instances of the same file open then all those instances would be restored. New behaviour is to only restore a single instance of the file in these cases.

**Report to:** peterfpeterson. 

**To test:**
- Open workbench and load multiple instances of one file.
- Close workbench and reopen it. Check only one instance of the file is restored.

Fixes #25004. 

---

#### Reviewer ####

Please comment on the fol [... cut]
closed 2019-03-06 by martyngigg