mantidproject/mantid #25014: Unexpected exception in project recovery

gemmaguest, 2019-03-05:
### Expected behavior
Project recovery should work

### Actual behavior
Project recovery throws an unexpected exception when attempting to recover

[image]

### Steps to reproduce the behavior
Follow the instructions in test1 in the unscripted testing. When trying to recover after crashing after the second, longer script, you get the exception

### Platforms affected
Windows 10 (bug doesn't occur on Ubuntu)

Pasarus, 2019-03-06:
Having found the cause, it is believed to be relatively minor and rare occurrence (this release may decide otherwise), it has been decided to not attempt to get it into the 4.0 release and push it back.

The cause is due to the limitations of OrderWorkspaceHistory and the accuracy of the system time that can be accessed from Windows and other Operating Systems themselves. i.e. RenameWorkspace runs too fast (Due to it being below 1 microsecond) to have a different start time than other RenameWorkspaces ran directly before or after it. This can ruin the OrderWorkspaceHistory's output ordering and thus make it not possible to recover.

Possible Solutions:
- If an algorithm's run time is less than a microsecond e.g. 0 on current OSs then sleep for 1 microsecond.
- Add the Algorithm ExecCount to the comment after the date and time on GeneratePythonScript and allow current ordering techniques to handle the rest.

gemmaguest, 2019-03-20:
I get the same problem for test 2 in the unscripted testing in workbench, but it is less severe because it catches the error - I just get a log message that the recovery fails.

PR #25283 (2019-03-21): Project Recovery: add execCount to the comment for call to GeneratePythonScript
**Description of work.**
Project recovery would in some cases be unable to recover because algorithms ran so quickly that the Operating system's time could not be rendered accurately enough to distinguish this fact. As added ordered attempts I have added the execCount to the comment section of script builds.

**To test:**
Perform the project recovery tests on a release build on a newly restarted PC.
http://developer.mantidproject.org/Testing/ErrorReporter-ProjectRecovery/ProjectRecoveryTesting.html

Fixes #25014 .  [... cut]
closed 2019-03-21 by martyngigg