mantidproject/mantid #25075: ACL data entry valdation

Mantid-Matthew, 2019-03-07:
I've run into the following issues most are minor but the first provokes an uncuaght exception :
 * If you enter some invalid values in the Grouping boxes such as negative numbers or letters you will get an uncaught exception, the GUI will then remain permanently disabled. 

 * The load button will not do anything if you have entered some invalid values with no indication of what is wrong. I.e. try selecting custom grouping and then not entering anything in the boxes
 * The GUI doesn't check that the default instrument is actually a Muon one
 * If you try to load a non-existent file or try to click the load button before the archive search is complete you will get an error message saying you have not specified a file. A more helpful error message would be to state that the file you have specified cannot be found.
 * For multiperiod data if you export results and then try and import results you will get an error message complaining that the loaded data has more than one spectrum
 * The tab ordering in the GUI appears to be random making navigating it a pain.

AnthonyLim23, 2019-03-08:
done the first bullet point in #25089

AnthonyLim23, 2021-03-01:
@Matt-Cumber have we fixed all of these in other issues?

Matt-Cumber, 2021-03-01:
> @Matt-Cumber have we fixed all of these in other issues?

Yes I believe so, I am not 100% sure what is meant by tab ordering but I see nothing random on the GUI

closed 2021-03-01 by AnthonyLim23