mantidproject/mantid #25080: Improvce release notes for elemental analysis v 2.0

NickDraper, 2019-03-08:
PR #24927 added some release notes, but I want a couple of improvements without holding up merging the PR.

Existing release notes text
````
- Elemental Analysis added to Muon Interfaces: Includes a selectable Periodic Table. There are two versions (Elemental Analysis and Elemental Analysis Old) and the difference is the plotting used.
```

Changes:
1. Add a screenshot of the new interface.
1. In the text we should state that the "old" interface is deprecated and may not be included in future releases.
1. Rather than just stating that the difference is the plotting used, could we not state why the new plotting is better?

AnthonyLim23, 2019-03-08:
we already had an issue #24918

Mantid-Matthew, 2019-03-08:
There is an open PR for improving the muon release notes which it may be worth piggy-backing onto here #25068. I have marked the PR as still in progress until the issues raised here are also addressed.

PR #25234 (2019-03-19): Elemental analysis and release notes
We have removed the original version of elemental analysis (it has never been in a release) at the request of the user.

**To test:**
Ask me for some data

Under Muon's check that there is only one elemental analysis
Open elemental analysis
Load the data - need to press the load button
When you click one of the detectors a plot will appear
Click some elements and lines should appear on the plot

Fixes #25080. 

---

#### Reviewer ####

Please comment on the following ([full description](http://developer.mantidproject.org/IndividualTicketTesting/)):

##### C [... cut]
closed 2019-03-20 by NickDraper