mantidproject/mantid #25122: Changing power scale on instrument view in workbench does not update view

martyngigg, 2019-03-12:
**Original reporter:** Steve King, ISIS. 

### Expected behavior

Changing power in power scale should update the instrument view.

### Actual behavior

If you are in Show Instrument view and you change the type of mapping to “power”, the colour scale updates but as you change the power value the instrument view itself doesn’t update.

### Steps to reproduce the behavior

* show the instrument;
* change the type of mapping to power
* the colour scale pdates but as you change the power value the instrument view itself doesn’t update.

If you enter a value a few too many times Mantid crashes…

### Platforms affected
All

martyngigg, 2019-03-12:
It turns out I have some time to look at this myself.

AndreiSavici, 2019-03-13:
Fixed in PR #25155

PR #25155 (2019-03-13): Fix power scale on workbench instrument view
**Description of work.**

Changing the value of the power on the power scale of the instrument view was broken at some point. This fixes it.

**Report to:** Steve King, ISIS 

**To test:**

Needs to be tested on both Workbench and MantidPlot.

* Load some data, e.g. `LOQ54431.raw` in the `SystemTestData`
* Show the instrument view
* Switch to the power scale
* Alter the power and confirm the changes alter the instrument view as well as the colour bar scale.
* Also confirm that editing the min/max values disables the autoscaling checkbox. 

Fixes #25 [... cut]
closed 2019-03-13 by AndreiSavici