mantidproject/mantid #25126: Minor fixes to Frequency domain analysis

Mantid-Matthew, 2019-03-12:
The following minor updates to the frequency domain analysis workflow have been requested.
 * For maxent change the x axis to read field in gauss rather than x axis
 * Change the x axis of FFT to field to match maxent
 * Set the default apodization to Lorenz and the default decay to 4.4
 * When you select some data with lots of small time bins followed by loading data with less time bins the large number of mzent data points is still enforced.

```
If I load a data file with lots of small time bins (e.g. HIFI 147218) it correctly insists I use a large number of data points in Maxent (values of 65536 upwards are offered). If I then switch back to other data with larger bins this limit is still enforced which makes Maxent run much more slowly than necessary.
```

AnthonyLim23, 2019-03-12:
https://github.com/mantidproject/mantid/blob/master/Framework/PythonInterface/plugins/algorithms/MuonMaxent.py#L440

AnthonyLim23, 2019-03-12:
the magic scale factor is `(1.e3/13.55)`

closed 2019-03-18 by martyngigg