mantidproject/mantid #25146: Can't open Workbench when switching between py2 and py3 builds

TTitcombe, 2019-03-13:
There is an error when switching to a py2 build and previously running `launch_mantidworkbench.sh` on a py3 build: `unsupported pickle protocol`. The manual fix is to delete `.config/mantidproject/mantidworkbench.ini`

### Steps to reproduce the behavior
Run `launch_mantidworkbench.sh` in a py3 build. Close and attempt to run the same script in a py2 build

### Platforms affected
Ubuntu (18)

marshallmcdonnell, 2019-03-13:
I actually ran into the exact same thing yesterday. In talking with @peterfpeterson, he suggested just a try-except line where it loads the state back in. 

I have a proposed fix in branch `25416_pickled_state`. Just testing before PR submission.

peterfpeterson, 2019-03-13:
Please create this as a PR for the [release](https://github.com/mantidproject/mantid/projects/17). Read: branch from `release-next`.

gemmaguest, 2019-03-14:
Fixed by PR #25154

PR #25154 (2019-03-14): Fix Workbench startup when switching Python versions
**Description of work.**

Adds try-except to skip a pickle issue between Python 2 and Python 3 for the cached state variable for the workbench. will default to behavior if not cached state was found.

**Report to:** @TTitcombe 

**To test:**
Build both python 2 and python 3 variants of the workbench. Open one and then the other, ensure no `ValueError: unsupported pickle protocol` is raised.

Fixes #25146
*This does not require release notes* because deals with non-User case of two workbench installs for different Python versions.

---

####  [... cut]
closed 2019-03-14 by gemmaguest