mantidproject/mantid #25192: Indirect data reduction symmetrise

AnthonyLim23, 2019-03-15:
Go to the interfaces and select data reduction from indirect
Change the instrument to IRIS
Enter run 26173
Change the detector grouping to individual 
Click Run

Go to the symmetrise tab
change it from file to workspace
Scroll to the bottom of the page and click run
It will give errors:
``` 
Error in execution of algorithm Symmetrise:
Failed to calculate array slice boundaries: index -1 is out of bounds for axis 0 with size 0
at line 87 in 'C:/MantidNightlyInstall/plugins/python/algorithms\Symmetrise.py'
Unknown error starting next batch algorithm
Got error from algorithm "Symmetrise"
Stopping batch algorithm because of execution error
```

SpencerHowells, 2019-03-15:
@AnthonyLim23 When data is loaded from file or workspace the limits in the Preview box are all zero. Move the vertical limit lines in the top plot & click on Preview. The Preview property values are now updated. Click on Run. Plot Result. I don't know if any user has ever run this!

Pasarus, 2019-03-15:
Having looked into it, @SpencerHowells is exactly right, this just seems like it needs a little bit better error handling telling a user what the issue is.

PR #25221 (2019-03-19): Indirect Data Reduction Interface, Symmetrise tab behaviour changes
**Description of work.**
Users were able to click the Run button before preview had been run, and preview was able to be ran before minimum or maximum, this causes errors because when preview is ran it sets some values for run. Therefore Run can only be clicked after preview has been, preview can only be clicked once the values for max or min has changed from their original values. This behaviour has been worked on between myself, @AnthonyLim23 and @SpencerHowells

**Report to:** Sanghamitra (ISIS)

**To test:**
- Change the i [... cut]
closed 2019-03-19 by soininen