mantidproject/mantid #25195: Workbench: Instrument View Crash

Pasarus, 2019-03-15:
### Expected behavior
Not crashing

### Actual behavior
Crashing

### Steps to reproduce the behavior
- Load INTER00013460 (Can be found in UnitTest data)
- Open instrumentview

### Platforms affected
Ubuntu 18.04

```
********* UNHANDLED EXCEPTION *********
  what(): Unable to normalize a zero length vector.

Backtrace:
 /home/sam/Builds/mantid1/bin/libMantidAPI.so(+0x34c931) [0x7fd551ff9931]
 /home/sam/Builds/mantid1/bin/libMantidAPI.so(+0x34ca55) [0x7fd551ff9a55]
 /usr/lib/x86_64-linux-gnu/libstdc++.so.6(+0x92a06) [0x7fd5597b2a06]
 /usr/lib/x86_64-linux-gnu/libstdc++.so.6(+0x92a41) [0x7fd5597b2a41]
 /usr/lib/x86_64-linux-gnu/libstdc++.so.6(+0x92c74) [0x7fd5597b2c74]
 /home/sam/Builds/mantid1/bin/libMantidKernel.so(_ZN6Mantid6Kernel3V3D9normalizeEv+0x72) [0x7fd55af25dea]
 /home/sam/Builds/mantid1/bin/libMantidQtWidgetsInstrumentViewQt5.so(+0x82fef) [0x7fd4cdcc2fef]
 /home/sam/Builds/mantid1/bin/libMantidQtWidgetsInstrumentViewQt5.so(+0x83446) [0x7fd4cdcc3446]
 /home/sam/Builds/mantid1/bin/libMantidQtWidgetsInstrumentViewQt5.so(_ZN8MantidQt13MantidWidgets20BankRenderingHelpers21renderRectangularBankERKN6Mantid8Geometry13ComponentInfoEm+0x30f) [0x7fd4cdcc5401]
 /home/sam/Builds/mantid1/bin/libMantidQtWidgetsInstrumentViewQt5.so(_ZN8MantidQt13MantidWidgets18InstrumentRenderer19drawRectangularBankEmb+0x2cc) [0x7fd4cdd3632e]
 /home/sam/Builds/mantid1/bin/libMantidQtWidgetsInstrumentViewQt5.so(_ZN8MantidQt13MantidWidgets18InstrumentRenderer4drawERKSt6vectorIbSaIbEEbb+0x159) [0x7fd4cdd3593b]
 /home/sam/Builds/mantid1/bin/libMantidQtWidgetsInstrumentViewQt5.so(_ZN8MantidQt13MantidWidgets18InstrumentRenderer16renderInstrumentERKSt6vectorIbSaIbEEbb+0x18a) [0x7fd4cdd35652]
 /home/sam/Builds/mantid1/bin/libMantidQtWidgetsInstrumentViewQt5.so(_ZNK8MantidQt13MantidWidgets15InstrumentActor4drawEb+0x4c) [0x7fd4cdcd66f6]
 /home/sam/Builds/mantid1/bin/libMantidQtWidgetsInstrumentViewQt5.so(+0x112356) [0x7fd4cdd52356]
 /home/sam/Builds/mantid1/bin/libMantidQtWidgetsInstrumentViewQt5.so(_ZNK8MantidQt13MantidWidgets17ProjectionSurface4drawEPNS0_14MantidGLWidgetEb+0xe0) [0x7fd4cdd55dd0]
 /home/sam/Builds/mantid1/bin/libMantidQtWidgetsInstrumentViewQt5.so(_ZNK8MantidQt13MantidWidgets17ProjectionSurface4drawEPNS0_14MantidGLWidgetE+0x6a) [0x7fd4cdd55cb0]
 /home/sam/Builds/mantid1/bin/libMantidQtWidgetsInstrumentViewQt5.so(+0xfea22) [0x7fd4cdd3ea22]
 /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5(_ZN7QWidget5eventEP6QEvent+0x1f8) [0x7fd539a0d038]
 /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5(_ZN19QApplicationPrivate13notify_helperEP7QObjectP6QEvent+0x9c) [0x7fd5399ce82c]
 /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5(_ZN12QApplication6notifyEP7QObjectP6QEvent+0x2a4) [0x7fd5399d60f4]
 /usr/lib/python2.7/dist-packages/PyQt5/QtWidgets.x86_64-linux-gnu.so(+0x3a9b4e) [0x7fd53a46ab4e]
 /usr/lib/x86_64-linux-gnu/libQt5Core.so.5(_ZN16QCoreApplication15notifyInternal2EP7QObjectP6QEvent+0x118) [0x7fd548a8b9a8]
 /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5(_ZN14QWidgetPrivate14sendPaintEventERK7QRegion+0x3a) [0x7fd539a0618a]
 /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5(_ZN14QWidgetPrivate10drawWidgetEP12QPaintDeviceRK7QRegionRK6QPointiP8QPainterP19QWidgetBackingStore+0x56a) [0x7fd539a0674a]
 /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5(_ZN14QWidgetPrivate11repaint_sysERK7QRegion+0x15b) [0x7fd5399dbf4b]
 /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x1afd2e) [0x7fd539a29d2e]
 /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x1b064d) [0x7fd539a2a64d]
 /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5(_ZN19QApplicationPrivate13notify_helperEP7QObjectP6QEvent+0x9c) [0x7fd5399ce82c]
 /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5(_ZN12QApplication6notifyEP7QObjectP6QEvent+0x2a4) [0x7fd5399d60f4]
 /usr/lib/python2.7/dist-packages/PyQt5/QtWidgets.x86_64-linux-gnu.so(+0x3a9b4e) [0x7fd53a46ab4e]
 /usr/lib/x86_64-linux-gnu/libQt5Core.so.5(_ZN16QCoreApplication15notifyInternal2EP7QObjectP6QEvent+0x118) [0x7fd548a8b9a8]
 /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5(_ZN22QGuiApplicationPrivate18processExposeEventEPN29QWindowSystemInterfacePrivate11ExposeEventE+0xcf) [0x7fd5461aa43f]```

soininen, 2019-03-15:
This may be fixed already in PR #25109.

martyngigg, 2019-03-15:
@mantid-roman

mantid-roman, 2019-03-15:
@soininen The latest changes in PR #25109 do not fix this issue. This is a worse case which may result in the component not drawn at all rather than just failing to calculate the surface normals.

PR #25245 (2019-03-21): Fix instrument view crash
**Description of work.**
Catch error and display a warning.

**To test:**
Open instrument view for INTER00013460

Fixes #25195 #25104

---

#### Reviewer ####

Please comment on the following ([full description](http://developer.mantidproject.org/IndividualTicketTesting/)):

##### Code Review #####

- Is the code of an acceptable quality?
- Does the code conform to the [coding standards](http://developer.mantidproject.org/Standards/)?
- Are the unit tests small and test the class in isolation?
- If there are changes in the release notes then do they describe the chang [... cut]
closed 2019-03-21 by martyngigg