mantidproject/mantid #25260: Indirect Data Analysis multiple files hard crash

Mantid-Matthew, 2019-03-19:
**Original reporter:** @Mantid-Matthew  

When Fitting multiple data in the Indirect Data Analysis interface if steps are taken in specific order mantid hard crashes.

### Steps to reproduce the behavior

Get the data from this PR #25250

1.Interfaces->Indirect->Data Analysis->ConvFit
2. Go to MultipleInput tab and click Add Workspace, add the first set of data then close the dialog reopen it and add the second set of data. Select All Spectra each time.
3. Select One Lorentzian as Fit type
4. Click Run and wait

### Platforms affected

robertapplin, 2019-03-19:
Instead of a hard crash, I get a runtime error:
[image]
This is on windows.

NickDraper, 2019-03-20:
Is there any progress on this?

NickDraper, 2019-03-20:
Actually the PR for this looks done and merged.  I'll close the issue, but reopen it if you disagree.

closed 2019-03-20 by NickDraper