mantidproject/mantid #25372: Workbench: Poor scaling with different monitor scaling in windows

NickDraper, 2019-03-25:
### Expected behavior
Workbench to appear "normal" 

### Actual behavior
On a system with a laptop and a monitor, it gets confused about scaling and just appears ugly.
Monitor: 1920*1080, scaling 100%
Laptop (Nick's Surface): 2736*1824, scaling 200%

Workbench on the Monitor
[image]

Workbench on the laptop
[image]

As a guess it appears to take the scaling from the laptop (primary screen), and uses it even though the window initially appears on the desktop.  Wehn you move the window to the laptop it gets even more confused.

If you launch it when the laptop it is not connected to the other screen it all looks normal.

### Steps to reproduce the behavior
See above 

### Platforms affected
Windows 10

NickDraper, 2019-03-26:
This might be helpful
https://stackoverflow.com/questions/39823918/how-to-approach-development-of-new-qt-5-7-high-dpi-per-monitor-dpi-aware-applic

hsautessella, 2019-07-11:
This has been (crudely) fixed by #26266 and a new issue #26301 has been raised with some updated information around the new issues that #26266 raises. Should we close this?

martyngigg, 2019-07-12:
I agree. Let's close this in favour of #26301

closed 2019-07-12 by martyngigg