mantidproject/mantid #25377: Remove scikit-image dependency

quantumsteve, 2019-03-25:
Switching to `ModestImage` in #24515 removed any direct calls to `scikit-image`, giving us the opportunity to remove it as a dependency.

I think skimage.transform.resize here could give additional options (for example the [documentation suggests using Gaussian smoothing when downsampling](http://scikit-image.org/docs/stable/auto_examples/transform/plot_rescale.html), but I haven't looked to see if it makes a significant difference in our colorfill plots or measured if it significantly slows down plotting?

quantumsteve, 2019-03-26:
Benefits of smoothing while resampling 
https://github.com/matplotlib/matplotlib/pull/13724

martyngigg, 2021-02-05:
This was removed in https://github.com/mantidproject/mantid/commit/83d8e4819458f78a2458e03f47ddd977b9da5c41.

closed 2021-02-05 by martyngigg