mantidproject/mantid #25379: Not loading ASCII files containing a footer

hortica, 2019-03-25:
Dr Anna Sokolova
ANSTO, BILBY instrument

### Expected behavior
I would assume that data files with a footer are legit.
For example, some commonly used software for protein scattering data analysis (ATSAS suit) creating files with footers.

### Actual behavior
When trying to load a file with a footer, the following error comes up:
"Error in execution of algorithm Load:
Failed to recognize this file as an ASCII file, cannot continue."

### Steps to reproduce the behavior
[test_footer.txt](https://github.com/mantidproject/mantid/files/3005840/test_footer.txt)
Please find file attached.
It is BILBY data with artificially added footer. With the footer is doesn't work, without it does.

### Platforms affected
I've checked it only on Windows 7 Enterprise, SP1.
Mantid release 1 March 2019
3.13.20190301.1751 (nightly build)

Mantid-Matthew, 2020-09-01:
@DavidFair I can't remember whether this was ever resolved David?

hortica, 2021-03-02:
With this, I am asking to keep the issue open.
I've tried the release 5.1.20210120.2323 and the ASCII files with footers still cannot be loaded.
I would prefer do not to specify a format for the footer; it can vary from file to file.
It would be good to have an algorithm that can read ASCII files and ignore the text footer, if its format is differ from the format of data lines.

The error message I am getting is the following:
"this file: x.dat, error description: bad lexical cast: source type value could not be interpreted as target
Error in execution of algorithm LoadAscii:
Failed to recognize this file as an ASCII file, cannot continue.
Error in execution of algorithm Load:
Failed to recognize this file as an ASCII file, cannot continue."

hortica, 2022-02-23:
The issue is still unresolved, and it would be great to have some comment/decision on it. Thank you in advance.

darominski, 2022-07-20:
If the lines of your footer started with a comment mark, by default `#`, then the error does not appear. Is it possible for you to produce footers that are not free text but contain a special character that would denote that it is a comment? It can be any character that you would then need to provide to the `CommentIndicator` property of the loader, unless you choose the default, then it's already taken care of.

The footer in this case is different than a header, where the comment indicator is not requested, but reading of the lines of the header are completely skipped when the data is loaded. It is harder to guess that a footer has started in the loader that is processing the file line by line, as it just appears to the code as an incorrect line: it is neither a comment, nor does it contain the numbers.

closed 2023-02-04 by stale[bot]