mantidproject/mantid #25383: Port ISIS Reflectometry GUI to Workbench

Pasarus, 2019-03-26:
Todo List:

- [ ] Sip exposure of highest level widget:
https://github.com/mantidproject/mantid/blob/isis_reflectometry_alternative_table/qt/scientific_interfaces/ISISReflectometry/GUI/MainWindow/MainWindowView.h#L27
- [ ] Python importing of exposed sip view widget and a presenter to go with the view.
- [ ] Qt5/Matplotlib plotting for workbench likely using MplCpp current way of doing this unknown discovery should be completed.
- [ ] "JobTreeView" plotting for MantidPlot/Qt4 isolated plot methods to work with original GUI in MantidPlot

Plot capability list:
- Over-plotting of all selected rows in the `JobTreeView`
- Workspaces plotted from the equivalent named in the ADS
- Use Log(x), Log(y) numbered axis for the plots of Reflectometry data by default

gemmaguest, 2019-03-28:
Here are some simple tests for various combinations of using the plot buttons.

### First, set up some processed rows
- Open the `ISIS Reflectometry` interface, select `INTER`.
- From the Reflectometry menu, select Import table and choose `INTER_NR_test2.tbl` from the ISIS sample data.
- Click `Process` and when finished expand all groups. All rows should now be green.

### Test the first button ("plot runs")
- Select a single row within a group and then click plot runs. You get a single curve:
[image]

- Select two rows and click plot runs. You get two curves on one plot:
[image]

- Select a group and click plot runs. You get all of the curves from the group on one plot:
[image]

- Select one row from the first group and two rows from the second group. You should get three curves:
[image]

- Select one row from the first group and the overall group for the second group. You should get three curves:
[image]

### Test the second button ("plot group")
- Select a row or rows within a group and click plot group. Nothing should happen (ideally this should be disabled).

- Select a group and click plot group. You should get a single curve (for the stitched result):
[image]

- Select two groups and click plot group. You should get two curves:
[image]

- Select one row from the first group and the overall group for the second group. The row should be ignored and one curve plotted for the second group:
[image]

gemmaguest, 2019-03-29:
Here is a suggested implementation for getting the selected workspace names for plotting. Feel free to adjust this if you think there is a better way.

- For groups, we want to plot the `Group::postprocessedWorkspaceName()` and for rows, `Row::reducedWorkspaceNames().iVsQBinned()`
- Create functions `RunsTable::selectedGroups()` and `::selectedRows()`, returning a `std::vector` of `Group` or `Row`.
- These should iterate through `m_selectedRowLocations`
- For each `RowLocation` get its `path()`. 
- The `Item`s are available via `reductionJobs()`:
  - Add functions to `ReductionJobs` to get a `Group` or `Row` from a `RowPath`.
  - Note that the path is a vector containing one int for a group (which is the group index in `m_groups`) and two ints for a row (the group index, and then the row index within the group - i.e. the index in `Group::rows()`).
  - We could consider making `RowPath` a proper class with some helper functions e.g. `isGroup()`, `isRow()`, `groupIndex()`, `rowIndexInGroup()`. These could make the calling code much more readable.
  - When requesting groups, ignore `RowPath`s that are not groups.
  - When requesting rows, if the `RowPath` is a group then iterate through its child rows and include them.
  - Note that rows are optional and should be excluded if not initialized (this represents an invalid row on the GUI).

PR #25605 (2019-06-04): Workbench: ISIS Reflectometry C++ GUI
**Description of work.**
This PR will allow users to access the reflectometry GUI from Workbench.

**To test:**
- Open Workbench
- Open Interfaces->Reflectometry->ISIS Reflectometry
- Run through the manual tests.
- Try and break the GUI running through random button clicking and the like.
- Check all buttons have a hint when hovered over.

Fixes #25383  

*Still need to add release notes*

---

#### Reviewer ####

Please comment on the following ([full description](http://developer.mantidproject.org/IndividualTicketTesting/)):

##### Code Review #####

-  [... cut]
closed 2019-06-05 by Pasarus