mantidproject/mantid #25508: Allow use of ISO646 complaint alternative operators

DanNixon, 2019-04-11:
We currently [disable use of alternative operators](https://github.com/mantidproject/mantid/blob/master/buildconfig/CMake/GNUSetup.cmake#L47). For some newer Boost libraries this is a problem.

All compilers should support these operators so there is little reason to explicitly disable them.

We probably want to stick to using the standard operators however.

SamJenkins1, 2019-06-28:
Simply removing the disabling from cmake fails on OSX, I would look into this further, but am leaving in half an hour, I've closed the PR for it.

DTasev, 2019-07-15:
I don't think that MSVC allows this at all, even on the latest version, unless we explicitly include a header to #define them.

I also can't find anything about Boost not behaving with `-fno-operator-names`, I am suspicious that they'd use them anyway. I've never seen those keywords used in C++ code, to be honest.

Overall I don't think we should remove the `-fno-operator-names` flag, especially since we would stick to using the standard operators, why allow it in the first place?

DanNixon, 2019-07-15:
`boost::sort` [uses them](https://github.com/boostorg/sort/blob/develop/include/boost/sort/block_indirect_sort/block_indirect_sort.hpp#L200).

DanNixon, 2019-07-25:
@DTasev Maybe you could also suggest that `boost::sort` moves away from these operators, I can't see much reason why they would prefer them over the standard ones (it seems pretty mixed too).

martyngigg, 2019-10-22:
Closing as not important enough.

closed 2019-10-22 by martyngigg