mantidproject/mantid #25534: SANS does not write dQ into 2D outputs

TTitcombe, 2019-04-15:
**Original reporter:** Steve King ISIS

Run any SANS reduction in 2D mode, select the 2D outputs and write to file. These output files do not contain dQ information.

1D outputs do contain this information, however the solution is not as simple as populating the 2D array by running the 1D dQ calculation

ricleal, 2019-04-15:
We have same problem at ORNL.
I looked into this and I don't see a easy way out... Even if the `Workspace2D` Y axis (`NumericAxis` / `BinEdgeAxis`) is modified to add `dQy`, it requires changes in `Rebin2D`. Besides, the `Workspace2D` seems hacked to contain 2D data. Qy/dQy can't contain bin edges... just centers.

TTitcombe, 2019-04-17:
@ricleal I think this was reported to me as a result of the issues ORNL were having, however I don't think ISIS will be looking into this for a little while

ricleal, 2019-04-17:
@TTitcombe Thanks for letting me know. We found a dirty temporary way of dealing with that: 3 workspaces (`I(Qx, Qy)`,  `dQx(Qx, Qy)`, and `dQy(Qx, Qy)`) grouped and that can be rebinned at the same time. I keep you update if you come up with cleaner way of doing this.