mantidproject/mantid #25550: Replace default matplotlib plot configuration dialog

martyngigg, 2019-04-16:
The out-of-the-box plot properties configuration dialog from matplotlib is hard to understand and doesn't include many options. This issue is to implement our own that will include the options in provides plus:
- the ability to modify legends
- remove curve from plots
- selectively enable/disable error bars for curves
- select colormap for image and colormesh plots

hsautessella, 2019-07-16:
My suggestions for what's now required from this issue since #25945 went in. Any feedback is welcome.

- A `Legend` tab for modifying legends.

[image]

- `Apply to all` buttons in the `Curves` sub-tabs (we may not want to apply the colours to each curve in these cases, this should be mentioned in the tool-tip).
- Icons displaying the colormaps in the colormap drop down menu (it's basically impossible to know what colormap names correspond to what colours).
- Make the text box in the `ColorSelector` widget editable; it's currently read-only. If you want to copy in an RGBA value you need to open the selector dialog and copy it in there, it would be better if we could copy it directly into the box. This would require some input validation.
- The `Remove` button on the curve selector can be a little big (see below). It would be nice to get this to the same height as the drop-down menu.

[image]

martyngigg, 2019-07-18:
@hsautessella This looks okay to me. I definitely agree with the icons for the colormaps.

What would be in the "Advanced Options" bit of the legend dialog? At some point we will also need to provide a way of editing legend entries more easily too.

We also need to balance this with the other work for the workbench for this release.

hsautessella, 2019-07-22:
The advanced legend options would look something like this: 
[image]

Users can currently edit legend entries by changing the labels of curves in the `Curves` tab, or is more direct legend editing required? Perhaps a double-click method similar to the axis editor, under the hood this could just set the curve labels?

I don't see any of the above as urgent so spreading the sub-issues over the coming releases sounds sensible depending on how much there is to do.

hsautessella, 2019-10-04:
All the issues in this epic are closed, so closing this as well

closed 2019-10-04 by hsautessella