mantidproject/mantid #25577: Workbench: Code commenting works incorrectly with inconsistent line endings

hsautessella, 2019-04-24:
### Expected behavior
Commenting code in a script with inconsistent line endings should work correctly.

### Actual behavior
Commenting code with inconsistent line endings will comment more lines than selected.

### Steps to reproduce the behavior
[test.zip](https://github.com/mantidproject/mantid/files/3111764/test.zip)
- Attached is a file with inconsistent line endings.
- Load the file into Workbench and try commenting some lines.
- Selecting a single line will comment multiple lines and selectling all lines will raise an error

martyngigg, 2019-09-02:
use a regex to find them

closed 2021-08-29 by stale[bot]

thomashampson, 2022-11-25:
Removing the stale label because I can't reproduce this on Linux or Windows.