mantidproject/mantid #25727: Normalisation of fit results in workbench

Harrietbrown, 2019-05-17:
### Expected behaviour
In workbench the fit parameters should correspond to the scaling on the plot
### Actual behaviour
The fit parameters automatically rescale to bin width, while the plot is not rescaled to bin width
### Steps to reproduce the behaviour
- Use Workbench
- From the sample data from the Mantid tutorial load GEM38370_Focussed.nxs
- Plot spectrum number 5
- set axis range to between `2` and `2.25`
- open fit
- add peaks corresponding to approximately `2.11`, `2.14` and `2.19`
- add linear background.
- press fit.
- Fit parameters have been normalised while the plot has not.
[image]
### Platforms affected
All.

AndreiSavici, 2019-05-17:
I think there is a deeper issue here. You are not plotting counts, you are plotting counts/Angstroms. And you don't know it, andyou cannot even change the preference in the workbench. I've change the preference in Mantidplot, and here is my image (with similar fit parameters)
[image]

hsautessella, 2019-05-17:
You're right about the units, I've been working on a PR that addresses this (#25556) and also another PR that will give the option to normalize/un-normalize on the plot's context menu (similar to the option in MantidPlot). 

I believe there is a plan to add a setting to not normalize by bin width by default, but I don't think an issue has been raised for it yet. In the mean time you can turn off normalizing by bin width by setting `config['graph1d.autodistribution'] = 'Off'`.

PR #26508 (2019-08-05): Normalisation of fit results on workbench
**Description of work.**
When fitting on Workbench, if the plot was normalised the fitting algorithm still ran on un-normalised data. The plotted line looked correct because line plotted using the fit parameters was normalised on plotting, but the fit results in the `FitPropertyBrowser` did not match the plot. This was due to the `m_shouldBeNormalised` flag not being set in Workbench as it is in MantidPlot.

This PR ensures the relevant flag is set before any fitting takes place.

**To test:**
- Follow the instructions in the associated issue and check [... cut]
closed 2019-08-05 by martyngigg