mantidproject/mantid #25728: Indirect Apply Absorption Corrections - use different names for MC and PP corrections

robertapplin, 2019-05-17:
**Original reporter:** Spencer

### Expected behavior
The corrected workspace which is output from the ApplyAbsorptionCorrections tab should not be overwritten if the corrections have come from a different method (i.e. PP and MC).

### Actual behavior
The corrected workspace is overwritten even when apply corrections which have come from a different method (i.e. PP or MC).

### Steps to reproduce the behavior
1. `Interfaces`->`Indirect`->`Corrections`->`Apply Absorption Corrections` tab
2. Load data (including corrections from MC)
3. Click `Run`
4. Load data (including corrections from PP)
5. Click `Run`
6. The original correct workspace is overwritten.

### Platforms affected
All

SpencerHowells, 2019-05-17:
@robertapplin Subtract & Apply - check logs have container_scale & shift and container filenames

robertapplin, 2019-07-02:
Fixed in #26045

closed 2019-07-02 by robertapplin