mantidproject/mantid #25776: muon frequency domain analysis FFT names output workspace name but

Harrietbrown, 2019-05-28:
### Expected behavior
output workspace from from transforms in muon frequency domain analysis should be named apropreatly so that they can be distinguished.

### Actual behavior
If a transform in performed without imaginary data, it is named in the same way as if it had be done with imaginary data using the last entry put in the imaginary workspace option.

### Steps to reproduce the behavior
1. open mantid plot
1. go to interfaces-->muon-->frequency domain analysis.
1. open EMU run 51341
1. go to transform tab
1. In workspace use `EMU51341; Pair Asym; long; \#1` and the same again in imaginary workspace
1. click Calcualte FFT
1. A new workspace as been created `FFT; Re EMU51341; PAir Asym; long; \#1; Im EMU51341; Pair Asym; long; \#1`. open and check contents.
1. uncheck Imaginary data and click click Calcualte FFT
1. no new workspace has been created, but `FFT; Re EMU51341; PAir Asym; long; \#1; Im EMU51341; Pair Asym; long; \#1` has been overwritten.

### Platforms affected

check with windows.

AnthonyLim23, 2019-07-04:
without the Im part the name should be:

`FFT; Re EMU51341; Pair Asym; long;`

AnthonyLim23, 2019-07-08:
+        print ("moo",imaginary_input_workspace )
+        Im_name = imaginary_input_workspace
+        if not self.view.imaginary_data:
+            Im_name = ""
+        fft_workspace_name = get_fft_workspace_name(input_workspace, Im_name)

PR #26267 (2019-07-15): Naming bug in frequency domain analysis
The FFT's in frequency domain analysis (FDA) would always add a name for the imaginary part of the FFT, even if one was not used. This fixes it.

**To test:**

Open FDA
Load some data
Go to the transform tab
Do a FFT
In the ADS a workspaces will have been added (its in a nested structure) that will reflect the names used for the real and imaginary parts. 
Untick the use imaginary part
Do an FFT
In the ADS a workspaces will have been added (its in a nested structure) that will reflect the names used for the real and imaginary parts (i.e. no imaginary part [... cut]
closed 2019-07-15 by NickDraper