mantidproject/mantid #25790: LoadDialog doesn't resize correctly

OwenArnold, 2019-05-29:
the `Load` algorithm dialog does not resize to show the full array of properties once the file (henc exact loader) has been selected. User has to resize dialog window themselves, which is annoying.

DanielMurphy22, 2020-04-28:
WB and Plot 5.0 seem to have the same behaviour now, which is they resize, maybe not to fully show the whole array or properties, but most.
 Anyhow, I think this is worth discussion so re-labelling.

martyngigg, 2020-12-17:
This appears to still be an issue. This is the Load dialog after selecting an `_event.nxs` file:
[image]

It would be good to try and give it some type of minimum size.

thomashampson, 2021-02-09:
Cannot reproduce the issue in Windows.

joseph-torsney, 2021-04-20:
Closing as we are not able to reproduce this

closed 2021-04-20 by joseph-torsney