mantidproject/mantid #25797: Workbench GUI mode

ElliotAOram, 2019-05-29:
### Expected behavior
* The workbench should be able to run `LoadDialog()` by default or inform users how to enable GUI mode to let them run this.

### Actual behavior
* By default running `LoadDialog()` causes:
```
RuntimeError: Can only display properties dialog in gui mode
  File "C:/Users/hvq65228/Documents/ESS presentations/load_script.py", line 18, in <module>
  File "C:\MantidInstall\bin\mantid\simpleapi.py", line 247, in LoadDialog
    set_properties_dialog(algm, **arguments)
  File "C:\MantidInstall\bin\mantid\simpleapi.py", line 1206, in set_properties_dialog
    raise RuntimeError("Can only display properties dialog in gui mode")
```

### Steps to reproduce the behavior
* Open the workbench
* Add the line `LoadDialog()` to the script editor
* Run it
* Observe the error produced in the message log

### Platforms affected
* All

NickDraper, 2019-09-02:
The underlying question here is are these used enough to justify their existence?

A GREP of the script Repo only finds 5 examples, 4 of which we created in the ExampleUser scripts.

The last appears to be an old script AnalyzeOneRun.py under SingleCrystal\Mantidplot_scripts.

@peterfpeterson @OwenArnold @gvardany Are these commands used at your facilities?  The ones where you add Dialog to the end of the algorithm name to launch the algorithm dialog window.  Would it affect any of you if these were dropped.

gvardany, 2019-09-03:
I don't think they are used at the ILL, at least for the normal operation that we recommend.

OwenArnold, 2019-09-03:
I'm not aware of these being used at the ESS, and I doubt these days that we would have any users who would want to do this gui-level scripting in the workbench.

peterfpeterson, 2019-09-03:
I don't know of them being used at ORNL either

DanielMurphy22, 2020-04-27:
I'm closing this as I believe it is settled that algorithm dialog commands won't be added and this should be handled with a documentation change to the Python in Mantid course to remove https://www.mantidproject.org/Algorithm_Dialogs

closed 2020-04-27 by DanielMurphy22