mantidproject/mantid #25842: Replace Boost::shared_ptr with std::shared_ptr

SamJenkins1, 2019-06-04:
`boost::shared_ptr` should be replaced with `std::shared_ptr` this will require work with boost python to allow use of `std::shared_ptr`

NickDraper, 2019-07-25:
What is the benefit of this that justifies the effort, and risk.
At least should we not just wait until we are using a version of Boost Python that natively understands std:shared_ptr

DanNixon, 2019-11-22:
> What is the benefit of this that justifies the effort, and risk.
> At least should we not just wait until we are using a version of Boost Python that natively understands std:shared_ptr

Benefit: moving away from 3rd party libraries when the standard will suffice
Effort: 1 hour and the ability to use a global find/replace tool
Risk: low

I see no reason this shouldn't be done.

martyngigg, 2020-03-05:
@giovannidisiena I'd advise coordinating with @Harrietbrown as she's doing the `#pragma once` replacement and you'll end up touching mostly the same files. I'd advise waiting until she's got her work merged.

Harrietbrown, 2020-03-05:
Mine will be done in sections by directories in `Framework`, #28200 has the checklist for when a chunk is merged in.

PR #28279 (2020-04-08): Replace boost::shared_ptr with std::shared_ptr
**Description of work.**
This PR replaces the use of `boost::shared_ptr` with `std::shared_ptr` and extends to other functions such as `boost::make_shared`, `boost::weak_ptr`, dynamic pointer casts and `boost::atomic`.

**To test:**
Check Mantid builds successfully, tests pass and the GUIs function normally.

Closes #25842 

*This does not require release notes* because it is an internal change related to code quality**

---

#### Reviewer ####

Please comment on the following ([full description](http://developer.mantidproject.org/ReviewingAPullRe [... cut]
closed 2020-04-08 by SimonHeybrock